espressif/pytest-embedded #44: TypeError: cannot pickle 'EncodedFile' object

DCSBL, 2022-01-11:
When running `pytest --port=/dev/cu.usbserial-01C3154E` with pytest-embedded-serial plugin, the following error message is thrown:

```
args(classes={'app': <class 'pytest_embedded.app.App'>, 'serial': <class 'pytes... object at 0x10ac2f5b0>}})), 'app': <pytest_embedded.app.App object at 0x10ac2f640>, 'gdb': None, 'openocd': None, ...}
_close_or_terminate = <function apply_count_generator.<locals>.wrapper.<locals>._close_or_terminate at 0x10ac50700>, res = []

    @functools.wraps(func)
    def wrapper(*args, **kwargs):
        def _close_or_terminate(obj):
            try:
                obj.close()
            except AttributeError:
                try:
                    obj.terminate()
                except AttributeError:
                    del obj
    
        res = []
        if COUNT == 1:
>           res = func(*args, **kwargs)

/.pyenv/versions/3.9.4/lib/python3.9/site-packages/pytest_embedded/plugin.py:150: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
/.pyenv/versions/3.9.4/lib/python3.9/site-packages/pytest_embedded/plugin.py:824: in dut
    return cls(**kwargs)
/.pyenv/versions/3.9.4/lib/python3.9/site-packages/pytest_embedded_serial/dut.py:27: in __init__
    self.serial.create_forward_io_process(self.pexpect_proc, source='serial')
/.pyenv/versions/3.9.4/lib/python3.9/site-packages/pytest_embedded/log.py:167: in create_forward_io_process
    self._forward_io_proc.start()
/.pyenv/versions/3.9.4/lib/python3.9/multiprocessing/process.py:121: in start
    self._popen = self._Popen(self)
/.pyenv/versions/3.9.4/lib/python3.9/multiprocessing/context.py:224: in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
/.pyenv/versions/3.9.4/lib/python3.9/multiprocessing/context.py:284: in _Popen
    return Popen(process_obj)
/.pyenv/versions/3.9.4/lib/python3.9/multiprocessing/popen_spawn_posix.py:32: in __init__
    super().__init__(process_obj)
/.pyenv/versions/3.9.4/lib/python3.9/multiprocessing/popen_fork.py:19: in __init__
    self._launch(process_obj)
/.pyenv/versions/3.9.4/lib/python3.9/multiprocessing/popen_spawn_posix.py:47: in _launch
    reduction.dump(process_obj, fp)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

obj = <Process name='Process-1' parent=30711 initial>, file = <_io.BytesIO object at 0x10ac4cc70>, protocol = None

    def dump(obj, file, protocol=None):
        '''Replacement for pickle.dump() using ForkingPickler.'''
>       ForkingPickler(file, protocol).dump(obj)
E       TypeError: cannot pickle 'EncodedFile' object

/.pyenv/versions/3.9.4/lib/python3.9/multiprocessing/reduction.py:60: TypeError
```

**Environment**:
OS: MacOS 12.0.1
Python version: Tested with 3.9 and 3.8.12
Connected hardware: esp32-wroom-32d
pytest.ini: 
```
[pytest]
addopts = --embedded-services serial

# log related
log_cli = True
log_cli_level = INFO
log_cli_format = %(asctime)s %(levelname)s %(message)s
log_cli_date_format = %Y-%m-%d %H:%M:%S

log_file = test.log
log_file_level = INFO
log_file_format = %(asctime)s %(levelname)s %(message)s
log_file_date_format = %Y-%m-%d %H:%M:%S
```

hfudev, 2022-01-11:
hi @DCSBL. I think it's already been fixed in https://github.com/espressif/pytest-embedded/pull/26. you can refer to that PR for more info.

could you have a try with a prerelease version by `pip install -U --pre pytest-embedded pytest-embedded-serial`?

DCSBL, 2022-01-11:
Nice. Yes that works. thanks.

closed 2022-01-11 by DCSBL