espressif/pytest-embedded #94: `plugin.py`: passing `--skip-autoflash` with no arguments results in no test collection

driftregion, 2022-06-09:
- `pytest --skip-autoflash` results in no test collection, **this should default to `True` or display an error**
- `pytest --skip-autoflash=y` skips autoflash and runs tests
- `pytest --skip-autoflash=n` runs tests
- `pytest` runs tests

hfudev, 2022-06-09:
@driftregion Hi, thanks for your report. I can't reproduce locally

```shell
ERROR: usage: pytest [options] [file_or_dir] [file_or_dir] [...]
pytest: error: argument --skip-autoflash: expected one argument
```

could you provide more info by answering these questions to help me debug on this?

1. what's your pytest-embedded version? if you're not using the latest version, could you try upgrading this?
2. are you running this with `--embedded-services esp,idf`? could you show a full shell output and your input?
3. are you using extra pytest.ini or extra conftest.py that may influence the output?

driftregion, 2022-06-10:
1. 
```
Name: pytest-embedded
Version: 0.7.3
```

2. yes, provided via `pytest.ini`

I just confirmed `examples/esp-idf/blink` does not have this problem. 

Closing as **cannot reproduce**