espressif/pytest-embedded #95: Change baud rade in embedded-serial service

KaDw, 2022-06-22:
Hi, 

Cool project, especially for integration testing on embedded systems.
I'm just checking it out but I'm not that familiar with python. How do I configure the baud rate for embedded-serial service?

hfudev, 2022-06-23:
Hi @KaDw ! Now you can use the environment variable `ESPBAUD` to specify it. Would improve the cli arguments in the future :) (#96)

KaDw, 2022-06-23:
One more question, I'm using Windows 10 and FT232 and CP210x UART-USB converter. 

At first I tried to use one serial converter with RX and TX shorted.
On the software side I'm basically using the echo example with slight modifications: I removed the conftest.py and modified test script as following:
```
import pytest

@pytest.mark.parametrize(
    'port',
    ['COM12'],
    indirect=True,
)
def test_echo_tcp(dut):
    dut.write('aaa')
    dut.expect('aaa')  # will decode automatically
```
This didn't work, `pexpect.expectations.TIMEOUT: Not found "aaa"`

Next just to make sure that it is not fault on the hardware side I prepared following setup: 

```
         RX <- TX    
COM7 <-> TX -> RX    <-> COM12
         GND <-> GND
```
The software side is the same.
I checked that I can see in COM12 everything that I type in COM7 and the other way 
Then every second I send 'a' character to COM7 which is received at COM12 (this is the DUT)

In both cases all I get is:
```
pexpect.expectations.TIMEOUT: Not found "aaa"
Bytes in curent buffer:
Please check the full lof here C:/Users/ ...
```

However, in COM7 I can see `aaa` so the dut.write executed correctly.

Any ideas what I might try to make the dut.expect (the RX side) work?

hfudev, 2022-06-23:
Unaware of why the first try failed, but for your second example, 
this line you're writing `dut.write('aaa')` is actually writing to the COM12 port, since the `dut` is using the `port` in your param.

This is a multi dut case, you may try this code block

```python
import pytest

@pytest.mark.parametrize(
    'count, port',
    [2, 'COM7|COM12'],
    indirect=True,
)
def test_echo_tcp(dut):
    com7 = dut[0] 
    com12 = dut[1]
    com7.write('aaa')
    com12.expect('aaa')  # will decode automatically

KaDw, 2022-06-23:
Good to know that the framework supports multiple duts!

About the second test. I'm writing using `dut.write('aaa')` to COM12 and see the output on COM7. This is not intended to be multi dut case but a case where I can see if the `write` method works fine. I ended up manually pushing 'a' over 30 seconds to make sure that the other side (COM12) can receive the characters.

I'm just trying to figure out why I can't receive anyting using `expect()` method and the buffer is empty. That's why I'm doing such weird things :P

hfudev, 2022-06-24:
dut.write (b'aaa') and dut.write ('aaa') should be the same.

> may be I should open another issue about that

sure. that would be better.

In the new issue, you may also attach a few additional info, like:
1. the cli command and the output
2. does it work with interacting with the port directly, without pytest-embedded? For example, can you receive bytes if using putty to send the characters?
3. `pip list | grep pytest-embedded`

and back to this issue, does the `ESPBAUD` workaround work? if it works you may close this issue and open the new one :)

KaDw, 2022-06-24:
The ESPBAUD workaround does not work.

I'm using Windows 10 and set the environmental variable in System variables (I also tried to set the env variable using `set ESPBAUD=921600`). I can see that it is set to correct value, `echo %ESPBAUD%` gives me 921600. When I call pytest the dut.write still uses 115200. I just used another usb-uart to montor the output of COM12 with 115200 and regardless of ESPBAUD I can only see correct characters when it is set to 115200.

igrr, 2022-06-24:
@hfudev i think that's probably because ESPBAUD is only taken into account in pytest-embedded-serial-esp _for flashing_, but not used in pytest-embedded-serial for the actual data exchange over serial.

igrr, 2022-06-24:
Maybe patching the value in Serial class can be a workaround for the time being. For example, in conftest.py:

```python
from pytest_embedded_serial.serial import Serial as ESerial
ESerial.DEFAULT_PORT_CONFIG['baudrate']=921600
```

KaDw, 2022-06-24:
@igrr Thanks, this workaround works!

closed 2022-06-24 by KaDw