espressif/pytest-embedded #97: Windows 10 echo example can't receive data

KaDw, 2022-06-24:
Hi,

I modified echo example a little bit to make it work with Windows 10. Since I can't really use socat so I used USB-UART adapter with shorted TX and RX connected to COM12. The problem is that the dut.write() is working but dut.expect() can't read anything.

I also verified that the same setup works on Ubuntu 20.04, I can both write and receive data.

Here is the test case:
```
import pytest

@pytest.mark.parametrize(
    'port',
    ['COM12'],
    indirect=True,
)
def test_echo_tcp(dut):
    dut.write(b'aaa')
    dut.expect('aaa')  # will decode automatically
```

pytest.ini:
```
[pytest]
addopts = --embedded-services serial

# log related
log_cli = True
log_cli_level = INFO
log_cli_format = %(asctime)s %(levelname)s %(message)s
log_cli_date_format = %Y-%m-%d %H:%M:%S
```

In addition to this changes I removed the conftest.py

Here is the output after I use `pytest` command in the test case directory:

```
================================================= test session starts =================================================

platform win32 -- Python 3.10.0, pytest-7.1.2, pluggy-1.0.0

rootdir: C:\rnd\pytest-rnd\first_test, configfile: pytest.ini

plugins: embedded-0.7.3

collected 1 item

 

test_echo.py::test_echo_tcp[COM12]

--------------------------------------------------- live log setup ----------------------------------------------------

2022-06-24 08:52:31 INFO Logs recorded under folder: C:\Users\user1\AppData\Local\Temp\pytest-embedded\2022-06-24_06-52-31\test_echo_tcp[COM12]

FAILED                                                                                                           [100%]

 

====================================================== FAILURES =======================================================

________________________________________________ test_echo_tcp[COM12] _________________________________________________

 

self = <pytest_embedded_serial.dut.SerialDut object at 0x00000168C441F1F0>, pattern = 'aaa', expect_all = False

args = (), kwargs = {}, patterns = ['aaa'], res = [], wrapped_buffer_bytes = ''

debug_str = 'Not found "aaa"\nBytes in current buffer: \nPlease check the full log here: C:\\Users\\user1\\AppData\\Local\\Temp\\pytest-embedded\\2022-06-24_06-52-31\\test_echo_tcp[COM12]\\dut.log'
[... 567 lines cut ...]
>               raise e.__class__(debug_str) from e

E               pexpect.exceptions.TIMEOUT: Not found "aaa"

E               Bytes in current buffer:

E               Please check the full log here: C:\Users\user1\AppData\Local\Temp\pytest-embedded\2022-06-24_06-52-31\test_echo_tcp[COM12]\dut.log

 

C:\Users\user1\AppData\Local\Programs\Python\Python310\lib\site-packages\pytest_embedded\dut.py:79: TIMEOUT

------------------------------------------------- Captured log setup --------------------------------------------------

INFO     root:dut.py:34 Logs recorded under folder: C:\Users\user1\AppData\Local\Temp\pytest-embedded\2022-06-24_06-52-31\test_echo_tcp[COM12]

=============================================== short test summary info ===============================================

FAILED test_echo.py::test_echo_tcp[COM12] - pexpect.exceptions.TIMEOUT: Not found "aaa"

================================================= 1 failed in 33.57s ==================================================

2022-06-24 08:52:31 aaa

```

hfudev, 2022-07-14:
@KaDw Hi, the `read_all` function is here: https://github.com/pyserial/pyserial/blob/master/serial/serialutil.py#L648

and the `serial.proc.write` in windows is here: https://github.com/pyserial/pyserial/blob/v3.5/serial/serialwin32.py#L303

I tried to debug for a while this morning, connected to an ESP-Prog with tx and rx shorted. Now I can reproduce your issue. Your test script works on Linux but fails on windows. Seems like the `write` method is working properly, no idea for now.

Will let you know if I found something! Thanks for reporting this!

hfudev, 2022-07-14:
@KaDw switching `read_all` to `read_until` works! could you check if it works on your side?

I'll create a PR and test for a while.

hfudev, 2022-07-14:
this unexpected behavior is related to: https://github.com/pyserial/pyserial/blob/v3.5/serial/serialwin32.py#L254

test with only `pyserial` code:

```python
from serial import Serial

serial = Serial('COM5')
serial.write(b'asdf\n')
print(serial.read_all())  # b''
print(serial.read_until())  # b'asdf\n')
```

hfudev, 2022-07-14:
more interesting stuff: this is also related to https://github.com/espressif/pytest-embedded/issues/61

after adding the `-s`, it works without any code change.

```ini
# pytest.ini
[pytest]
addopts =  --embedded-services serial -s`
```

KaDw, 2022-07-14:
> more interesting stuff: this is also related to #61
> 
> after adding the `-s`, it works without any code change.
> 
> ```ini
> # pytest.ini
> [pytest]
> addopts =  --embedded-services serial -s`
> ```

Amazing! This works.
I guess the code change is not required then.

hfudev, 2022-07-14:
glad it works! I'll dig deeper into it later. guess it's related to `sys.stdout`, `pytest`, and windows. Shall we close this issue? I'll track it on #61.

KaDw, 2022-07-14:
We can close it, btw what is the `-s` flag doing? is it pytest flag `  -s                    shortcut for --capture=no.`?

KaDw, 2022-07-14:
I'm just thinking, Should we add a comment or even the -s option to one of the pytest.ini example for now? Hopefully it will also help people using windows.

hfudev, 2022-07-14:
yes. it is. by default the pytest will capture the sys.stdout. Didn't look into their code, but seems like somehow it's related...

> Should we add a comment or even the -s option to one of the pytest.ini example for now?

I'll add this to "known limitation" chapter in the documentation.