espressif/pytest-embedded #168: Qemu dut frequently locked after 'Press ENTER to see the list of tests'

zikalino, 2022-11-29:
I have observed following problem when running unity tests on QEMU:

    dut.expect_exact('Press ENTER to see the list of tests')
    dut.write('[qemu]')
    dut.expect_unity_test_output(timeout=300)

After sending "[qemu]" to dut, there's no response from dut. This happens around 25% of time.

I have tried to run test application directly as follows:

    qemu-system-xtensa -nographic -no-reboot -machine esp32 -drive file=/esp-idf/components/esp_ringbuf/test_apps/build_esp32_default/flash_image.bin,if=mtd,format=raw

and this problem can't be reproduced -- dut is always responding correctly.

zikalino, 2022-11-29:
**UPDATE**

Apparently slight delay before writing to **dut** helps. I have used following code to display list of tests and then continue with requesting test execution and it seems to work flawlessly:

    dut.expect_exact('Press ENTER to see the list of tests')
    sleep(1)
    dut.write('\n')
    sleep(1)
    dut.write('[qemu]')
    dut.expect_unity_test_output(timeout=300)

hfudev, 2022-11-30:
Hi @zikalino, could you provide more info about your running environment? like which os are you using? which terminal are you using?

closed 2023-05-31 by hfudev