espressif/pytest-embedded #185: pytest-embedded-serial leaves files open after test

JammuKekkonen, 2023-01-26:
Running tests using pytest-embedded-serial, eventually either OS or python runs out of file handles as the framework seems to leave open file handles after test execution ends. Attached is a example case with three duts, to me running this fails systematically at test 167 and beyond with
> ValueError: filedescriptor out of range in select()

Note: might need to increase OS file handle limit to hit Python's limit above, other option is that tests start to fail with
> OSError: [Errno 24] Too many open files

[test_open_files.zip](https://github.com/espressif/pytest-embedded/files/10508657/test_open_files.zip)

hfudev, 2023-01-28:
@JammuKekkonen Thanks for reporting this issue. I can reproduce locally. I'll take a look.

closed 2023-01-29 by hfudev

hfudev, 2023-01-29:
@JammuKekkonen the bugfix version 1.2.2 has been published. Please have a try.

JammuKekkonen, 2023-01-30:
@hfudev I can confirm that bug is fixed in 1.2.2.
Thanks for the quick reply and fix!

PR #188 (2023-01-29): Fix/leaking filedescriptors
closes #185