espressif/pytest-embedded #199: pytest-embedded 1.2.4 break qemu serial option set in pytest.ini

darthcloud, 2023-03-11:
I noticed my test suite failing to launch this morning without any change made in my repo.

Cause look to be the 1.2.4 update to pytest-embedded / qemu

I set the qemu serial option from my pytest.ini file:

```
addopts = --embedded-services idf,qemu -s --qemu-image-path ./build/flash_image.bin --qemu-extra-args '"-serial tcp::5555,server,nowait"'
```
Full file here: https://github.com/darthcloud/BlueRetro/blob/master/pytest.ini

Used to work fine with 1.2.3

Now in 1.2.4 I get following error:
```
2023-03-11 18:18:41 INFO Executing qemu-system-xtensa -nographic -no-reboot -machine esp32 -serial tcp::5555,server,nowait -drive file=./build/flash_image.bin,if=mtd,format=raw
2023-03-11 18:18:41 qemu-system-xtensa: -serial tcp::5555,server,nowait: invalid option
ERROR
```

Full log previous run on 1.2.3:
https://github.com/darthcloud/BlueRetro/actions/runs/4349022457/jobs/7598200637

Full log run on 1.2.4:
https://github.com/darthcloud/BlueRetro/actions/runs/4393563548/jobs/7694043699

darthcloud, 2023-03-11:
On my local machine upgrading to 1.2.4 break qemu launch too.

After forcing a rollback with:
```
pip install --force-reinstall -v "pytest-embedded-qemu==1.2.3"
```

It work again.

igrr, 2023-03-11:
Could you please try `--qemu-extra-args '-serial tcp::5555,server,nowait'` instead of `--qemu-extra-args '"-serial tcp::5555,server,nowait"'`? Otherwise "-serial tcp:5555,server,nowait" is passed to QEMU as a single argument, so QEMU can't recognize it.

(Previously this happened to work because we were launching QEMU via the shell, which tried to split the arguments.)

darthcloud, 2023-03-11:
This make pytest throw an error:
```
ERROR: usage: pytest [options] [file_or_dir] [file_or_dir] [...]
pytest: error: argument -s: ignored explicit argument 'erial tcp::5555,server,nowait' (via addopts config)
```

without the double quote it look to be taken as a pytest argument

addopts line used:
```
addopts = --maxfail=1 --embedded-services idf,qemu -s --qemu-image-path ./build/flash_image.bin --qemu-extra-args '-serial tcp::5555,server,nowait'
```

hfudev, 2023-03-13:
@darthcloud I've reproduced your issue locally. After some investigation, I think this is a pytest bug.

### Reproduce Steps

```ini
[pytest]
addopts = --qemu-extra-args "-serial tcp::5555,server,nowait"
```

pytest is using using [iniconfig](https://github.com/pytest-dev/iniconfig) to parse the ini file and `shelx.split()` to parse the argument. [code link](#https://github.com/pytest-dev/pytest/blob/7af1e4e4edf706c261a246b3a060cf7b906f617b/src/_pytest/config/__init__.py#L1482)

Here's a simpler version of their code to parse the value from the ini file:

```python
ini = iniconfig.IniConfig('pytest.ini')
# the value here is still correct: --qemu-extra-args "-serial tcp::5555,server,nowait"
value = ini['pytest']['addopts']
print(shlex.split(value) if isinstance(value, str) else value)
# this line is wrong. it would return ['--qemu-extra-args', '-serial tcp:5555,server,nowait']
```

if use `shlex.split(value, posix=False)`, the final result will be as expected: `['--qemu-extra-args', '"-serial tcp::5555,server,nowait"']`

btw, passing these args via CLI will also go into the same issue.

### Workaround

you may use parametrization instead.
```python
        @pytest.mark.parametrize('qemu_extra_args', [
            '"-serial tcp::5555,server,nowait"',
        ], indirect=True)
        def test_hid_keyboard_descriptor(blueretro):
            ...
```

hfudev, 2023-03-13:
Besides, this `posix=True` technique could also be improved on our side, though it would make the CLI call less intuitive.

```shell
pytest --qemu-extra-args '"-serial ..."'
```

darthcloud, 2023-03-14:
Thanks @hfudev

Ultimately I decided to simply launch QEMU with my GitHub action before running pytest.

I'll let you decide if you want to close this or try to handle it within pytest-embeded.

hfudev, 2023-03-23:
@darthcloud Cool. Glad it works.

It would be a simple fix, but kinda dirty... I'll close this issue for now. Thanks for reporting :)

closed 2023-03-23 by hfudev