espressif/pytest-embedded #200: Writing to DUT doesn't work when application console is configured with ESP_CONSOLE_USB_SERIAL_JTAG

robbawebba, 2023-03-21:
I observed a difference in behavior with writing to the the serial connection with the DUT when the running application uses the USB Serial/JTAG interface for ESP console, rather than UART (the default settings). It seems that `dut.expect` works regardless of the ESP console interface, but `dut.write` only seems to work when UART is used for the ESP console.

The Kconfig options I enabled in the application to reproduce this are as follows:
```
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
```

hfudev, 2023-03-22:
@robbawebba could you share your c code? I'll see if I can reproduce it.

robbawebba, 2023-03-22:
@hfudev Sure, tomorrow I can try to reproduce this with the IDF console component examples. If that doesn't work, I'll have to come up with a different minimal application to reproduce it (or I'll discover that it's an issue with my existing application :sweat_smile:)

robbawebba, 2023-03-22:
@hfudev Below is a minimal example app to help reproduce the issue. My application (and this example) uses the convenience functions from the `console` component to initialize the REPL and linenoise library. This is different from the examples, where the libraries are initialized directly by the application.

Please let me know if you think this issue is more appropriate for the IDF repository, or if this is not an SDK issue and should be discussed in the forums.

App C code:
```
#include <stdio.h>
#include <string.h>
#include <fcntl.h>
#include "esp_system.h"
#include "esp_log.h"
#include "esp_console.h"
#include "linenoise/linenoise.h"
#include "argtable3/argtable3.h"
#include "cmd_nvs.h"
#include "cmd_system.h"
#include "esp_vfs_cdcacm.h"
#include "nvs.h"
#include "nvs_flash.h"

static void initialize_nvs(void)
{
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK(err);
}

void app_main(void)
{
    esp_console_repl_t *repl = NULL;
    esp_console_repl_config_t repl_config = ESP_CONSOLE_REPL_CONFIG_DEFAULT();

    repl_config.prompt = CONFIG_IDF_TARGET "> ";

    initialize_nvs();

    // setup REPL with the configured interface
#if CONFIG_ESP_CONSOLE_UART
    esp_console_dev_uart_config_t uart_config = ESP_CONSOLE_DEV_UART_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_console_new_repl_uart(&uart_config, &repl_config, &repl));
#elif CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG
    esp_console_dev_usb_serial_jtag_config_t usb_jtag_config = ESP_CONSOLE_DEV_USB_SERIAL_JTAG_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_console_new_repl_usb_serial_jtag(&usb_jtag_config, &repl_config, &repl));
#endif

    /* Register commands */
    esp_console_register_help_command();
    register_system_common();
    register_system_sleep();
    register_nvs();

    /* Run the REPL task */
    ESP_ERROR_CHECK(esp_console_start_repl(repl));
}
```

sdkconfig.defaults:

hfudev, 2023-03-23:
Hi @robbawebba, unfortunately I can't reproduce your error. Since I don't have the `cmd_nvs.h` and `cmd_system.h`, I modified the source code a bit.

### C File

```c
#include <stdio.h>
#include <string.h>
#include <fcntl.h>
#include "esp_system.h"
#include "esp_log.h"
#include "esp_console.h"
#include "linenoise/linenoise.h"
#include "argtable3/argtable3.h"
#include "esp_vfs_cdcacm.h"
#include "nvs.h"
#include "nvs_flash.h"

static void initialize_nvs(void)
{
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK(err);
}

void app_main(void)
{
    esp_console_repl_t *repl = NULL;
    esp_console_repl_config_t repl_config = ESP_CONSOLE_REPL_CONFIG_DEFAULT();

    repl_config.prompt = CONFIG_IDF_TARGET "> ";

    initialize_nvs();

    // setup REPL with the configured interface
#if CONFIG_ESP_CONSOLE_UART
    esp_console_dev_uart_config_t uart_config = ESP_CONSOLE_DEV_UART_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_console_new_repl_uart(&uart_config, &repl_config, &repl));
#elif CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG
    esp_console_dev_usb_serial_jtag_config_t usb_jtag_config = ESP_CONSOLE_DEV_USB_SERIAL_JTAG_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_console_new_repl_usb_serial_jtag(&usb_jtag_config, &repl_config, &repl));
#endif

    /* Register commands */
    esp_console_register_help_command();

    /* Run the REPL task */
    ESP_ERROR_CHECK(esp_console_start_repl(repl));
}
```

### pytest file

modified a bit as well. (expect something then start writing is more stable than `time.sleep`)

```python
from pytest_embedded import Dut

def test_console_help(dut: Dut) -> None:
    dut.expect('esp32s3>')
    dut.write('help')
    # expect some known string from the help output
    dut.expect_exact('Print the list of registered commands')
```

### Build & Test Process

```shell
[... 54 lines cut ...]
2023-03-23 11:12:38 I (103) spi_flash: flash io: dio
2023-03-23 11:12:38 W (103) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
2023-03-23 11:12:38 I (104) sleep: Configure to isolate all GPIO pins in sleep state
2023-03-23 11:12:38 I (104) sleep: Enable automatic switching of GPIO sleep configuration
2023-03-23 11:12:38 I (105) app_start: Starting scheduler on CPU0
2023-03-23 11:12:38 I (105) app_start: Starting scheduler on CPU1
2023-03-23 11:12:38 I (105) main_task: Started on CPU0
2023-03-23 11:12:38 I (115) main_task: Calling app_main()
2023-03-23 11:12:38 ^[[0n
2023-03-23 11:12:38 Type 'help' to get the list of commands.
2023-03-23 11:12:38 Use UP/DOWN arrows to navigate through command history.
2023-03-23 11:12:38 Press TAB when typing command name to auto-complete.
2023-03-23 11:12:38 
2023-03-23 11:12:38 Your terminal application does not support escape sequences.
2023-03-23 11:12:38 
2023-03-23 11:12:38 Line editing and history features are disabled.
2023-03-23 11:12:38 
2023-03-23 11:12:38 On Windows, try using Putty instead.
2023-03-23 11:12:38 esp32s3>  I (425) main_task: Returned from app_main()
2023-03-23 11:12:38 help
2023-03-23 11:12:38 help 
2023-03-23 11:12:38   Print the list of registered commands
2023-03-23 11:12:38 
2023-03-23 11:12:38 esp32s3>  .
```

robbawebba, 2023-03-23:
Hi @hfudev, thank you for your time and effort in trying to reproduce this! I reviewed your logs and noticed several differences in your development/testing environment compared to mine. I tried to match your environment as close as possible, including python3 version, pytest plugin versions, etc.

It turns out that the IDF version was the key difference in our development environments. I have been using IDF v4.4.1 (the IDF version that my product firmware is using right now, until one of our dependent releases receives v5.0 support), whereas you are using IDF v5.1+. I built the application using the latest Docker image (espressif/idf:latest) and was able to successfully run this test.

So this turned out to be not an issue with pytest-embedded, rather an issue with ESP-IDF in v4.4.1 that is fixed in some later release. I searched the IDF issue tracker for reports of serial issues when using CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG, and there seem to be some issues and fixes that have been released since v4.4.1.

So I'll close this issue. Thank you for your help

closed 2023-03-23 by robbawebba