espressif/pytest-embedded #217: Could the timeout for dut.expect()  exceed 300s?

feizi, 2023-06-16:
When I set timeout=16000s or timeout=None in `dut.expect(pattern, timeout=timeout)`, I get the following error.   
Is there any way to increase the timeout to 16000s?
```
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++ Timeout +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ Stack of Thread-1 (_event_loop) (140579299362560) ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  File "/home/sunxiangyu/anaconda3/lib/python3.10/threading.py", line 973, in _bootstrap
    self._bootstrap_inner()
  File "/home/sunxiangyu/anaconda3/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/home/sunxiangyu/anaconda3/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/home/sunxiangyu/.espressif/python_env/idf5.0_py3.10_env/lib/python3.10/site-packages/pytest_embedded_serial/serial.py", line 194, in _event_loop
    time.sleep(0.05)  # set interval

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ Stack of QueueFeederThread (140579307755264) ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  File "/home/sunxiangyu/anaconda3/lib/python3.10/threading.py", line 973, in _bootstrap
    self._bootstrap_inner()
  File "/home/sunxiangyu/anaconda3/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/home/sunxiangyu/anaconda3/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/home/sunxiangyu/anaconda3/lib/python3.10/multiprocessing/queues.py", line 231, in _feed
    nwait()
  File "/home/sunxiangyu/anaconda3/lib/python3.10/threading.py", line 320, in wait
    waiter.acquire()

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++ Timeout +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
FAILED

E   Failed: Timeout >300.0s
```

hfudev, 2023-06-17:
Hi @feizi! I can't reproduce locally with the following script.

```python
def test_expect_long_timeout(dut):
    dut.write('this would be redirected')

    dut.expect('foo bar', timeout=600)
```

The error message looks good.

```
>               raise e.__class__(debug_str) from e
E               pexpect.exceptions.TIMEOUT: Not found "foo bar"
E               Bytes in current buffer (color code eliminated): this would be redirected
E               Please check the full log here: /tmp/pytest-embedded/2023-06-16_13-57-47-249527/test_expect_long_timeout/dut.log

/home/fuhanxi/esp/pytest-embedded/venv/lib/python3.11/site-packages/pytest_embedded/dut.py:82: TIMEOUT
===================================================================================================== short test summary info ======================================================================================================
FAILED test.py::test_expect_long_timeout - pexpect.exceptions.TIMEOUT: Not found "foo bar"
================================================================================================== 1 failed in 600.14s (0:10:00) ===================================================================================================
```

Could you help provide more details?
- the output of `pip list`
- the test script you're using
- the full log after you execute `pytest`

hfudev, 2023-06-19:
@feizi Glad the issue is solved. I'll close this issue.

closed 2023-06-19 by hfudev