espressif/pytest-embedded #224: one process - one dut

horw, 2023-08-10:
**Introdution**
I suggest main problem why pytest-embedded can't use multiprocessing rn lies in the fact that dut object init in the main process, and after init we try to pass this object to another process, there are not clear usage of object replica.

**Related**
This problem have same root like issue #162 

**Thoughts**
Pytest-embedded is a test framework, and maybe its possible separate responsibility between **test** and **controlling device** tasks.<br/>
Pytest-embedded can implement main process like a publisher (which only responsible for tasks and results, its don't worry about dut itself). Dut-control processes are subscribers, when one of them receive commands, it start work around how to run this test cast, this mean publisher delegate this responsibility. Every dut init in this dut-control processes inside, through main process collect class parameters and send them to dut-control processes. Maybe here we also can delegate responsibility of collection parameters to dut-control, and just send needed console args parameters.
<br/>
<br/>
schema:
[image]

**Summary**
In this way pytest-embedded objects responsibility will be more clear. And it will help pytest-embedded framework achieve better result in task parallelism.

hfudev, 2023-08-10:
related #162

horw, 2023-09-01:
It not possible to implement it, because need refactor all methods and api. Also have some issue with support: on one side need update api, and on another side need update client logic(dut), double work.

closed 2023-09-01 by horw