espressif/pytest-embedded #226: Windows 10 Not found Match

breakersun, 2023-08-26:
**Describe the bug**
On Windows 10

**Configuration Files**
- [ ] `pytest.ini`
```
[pytest]
addopts = --embedded-services serial -s

# log related
log_cli = True
log_cli_level = INFO
log_cli_format = %(asctime)s %(levelname)s %(message)s
log_cli_date_format = %Y-%m-%d %H:%M:%S
```

**To Reproduce**
Steps to reproduce the behavior:
1. Run `pytest -v`
2. Got error message 
``` 
>               raise e.__class__(debug_str) from e
E               pexpect.exceptions.TIMEOUT: Not found "[ATSS]:ATS"
E               Bytes in current buffer (color code eliminated): # [ATSS]:ATS:START
E               Please check the full log here: C:\Users\403283\AppData\Local\Temp\pytest-embedded\2023-08-26_05-55-34-035987\test_echo_tcp[COM4]\dut.log

..\..\scoop\apps\python\current\Lib\site-packages\pytest_embedded\dut.py:82: TIMEOUT
=========================== short test summary info ===========================
FAILED test_ats.py::test_echo_tcp[COM4] - pexpect.exceptions.TIMEOUT: Not fou...
============================== 1 failed in 1.66s ==============================
```

**Expected behavior**
It should match the pattern in the buffer. the pattern actually in the buffer.

**Dev Environment (please complete the following information):**
 - OS: Windows 10
 - Python Version 3.11.1
 - Pytest Version 7.4.0
 - pytest-embedded Version 1.3.4

breakersun, 2023-08-26:
my test is pretty simple

```python
import pytest

@pytest.mark.parametrize( 'port', ['COM11'], indirect=True,)

def test_echo_tcp(dut):
    dut.write(b'ATSS=START\r\n')
    dut.expect('[ATSS]:ATS', timeout=1)

```

DUT log is like this:
```

# [ATSS]:ATS:START

```

breakersun, 2023-08-26:
I also tried with ubuntu, and got the same error.

OS: Ubuntu 18.04
Python Version 3.10
Pytest Version 7.4.0
pytest-embedded Version 1.3.4

igrr, 2023-08-26:
Could you try `dut.expect_raw`? Otherwise the match string is handled as a regular expression, and due to the square brackets it will have different meaning than the string you are trying to match.

hfudev, 2023-08-27:
> Could you try `dut.expect_raw`? Otherwise the match string is handled as a regular expression, and due to the square brackets it will have different meaning than the string you are trying to match.

@breakersun Please try `dut.expect_exact` instead. The details and difference between `expect` and `expect_exact` are explained [here](https://docs.espressif.com/projects/pytest-embedded/en/latest/expecting.html#expect-unity-test-output-kwargs)

breakersun, 2023-09-05:
Thank you @hfudev dut.expect_exact works

closed 2023-09-05 by breakersun