espressif/pytest-embedded #232: Pytest-embedded leaks file descriptors when used with pytest-lazy-fixture

JammuKekkonen, 2023-10-10:
**Describe the bug**
Running pytest-embedded-serial and dividing the serial lines using fixtures and pytest-lazy-fixture leaves files open so eventually either OS or python runs out of file descriptors.

I have included a test example that reproduces the problem. Essentially I have a multi-dut test system where I detect and divide the connected duts into single dut references using fixtures. I can then run same tests to multiple duts or dut combinations by parametrizing them, but that requires lazy-fixtures.

It looks like at least one issue is with pytest_embedded_serial/serial.py, for some reason the multiprocessing.MessageQueue is not empty when 'end' command is processed so it's left hanging. I have attached a patch file that for me improves the situation somewhat so I can run more tests before eventual failure. There seems to be some other place as well that leaks files.

**To Reproduce**
Steps to reproduce the behavior:
1. unzip test_alot.zip
2. install pytest-lazy-fixture
3. Run `pytest ...`

**Expected behavior**
Tests should be able to run, test framework should clear all open files properly.

**Dev Environment (please complete the following information):**
 - OS: Debian Linux
 - Python Version: 3.9.2
 - Pytest Version 7.4.2
 - pytest-embedded Version 1.3.5
 - pytest-lazy-fixture Version: 0.6.3

[improvement.patch](https://github.com/espressif/pytest-embedded/files/12857476/improvement.patch)
[test_alot.zip](https://github.com/espressif/pytest-embedded/files/12857477/test_alot.zip)

hfudev, 2023-10-11:
Hi @JammuKekkonen! Thank you for reporting this issue and the patch! I'll take a look.

horw, 2023-10-12:
Hi @JammuKekkonen, I just tried to reproduce this error with these parameters:
```-s --count 2 --embedded-services esp,idf  --skip-autoflash y --app-path /home/horw/PycharmProjects/pytest-embedded/tests/fixtures/unit_test_app_esp32c3 ```
I ran the test, which was in test_alot.zip, and it just passed(without patching).
Can you point out what I missed to reproduce this error?
btw, I ran it on Ubuntu.
[image]

JammuKekkonen, 2023-10-12:
Hi @horw , my example is using serial lines to communicate with dut. I included the pytest.ini file too in the zip. I'm just running pytest without extra options, but serial communication is set up in the ini.

horw, 2023-10-23:
> @hfudev @horw Any luck reproducing the problem? Let me know if you need more info or if I can assist with testing patches.

Hi, sorry for the delayed response. I will check it tomorrow.

horw, 2023-10-24:
@JammuKekkonen, I can reproduce your bug locally and have created PR #234 to fix it. However, in my case, the patch you provided would make the test script fail at around the 500th iteration. Without the patch, my failures started after 250+ tests. Here is my test results [Test Results.log](https://github.com/espressif/pytest-embedded/files/13112851/Test.Results.log).

horw, 2023-10-25:
Today, I checked which file descriptors pyserial, msg_query, etc., opened and closed in pytest-embedded. It seems like pyexpect may have some issues.
I debugged the code with the following script:
```python
def get_fd():
    import os
    fls = []
    # You can use listdir without readlink, but then there will be one 'fake' file descriptor, which 'proc/self/fd' opens by itself.
    for f in os.listdir('/proc/self/fd'):
        try:
            link = os.readlink(f'/proc/self/fd/{f}')
            fls.append(f)
        except Exception as e:
            pass
   return fls, len(fls)
print(get_fd())
```

#  The status of file descriptor (fd) opening and closing
## init stage
['0', '1', '2', '3', '4', '5'] 6 - total # start program

['0', '1', '2', '3', '4', '5', '6', '7', '8', '9'] 10 # init two msqquery

['0', '1', '2', '3', '4', '5', '6', '7', '8', '9', '10', '11', '13'] 13 # init first pexpect

['0', '1', '2', '3', '4', '5', '6', '7', '8', '9', '10', '11', '12', '13', '14', '16'] 16 # init second pexpect

['0', '1', '2', '3', '4', '5', '6', '7', '8', '9', '10', '11', '12', '13', '14', '15', '16', '17', '18', '19', '20', '21', '22'] 23 # after init first Serial instance (pyserial)

['0', '1', '2', '3', '4', '5', '6', '7', '8', '9', '10', '11', '12', '13', '14', '15', '16', '17', '18', '19', '20', '21', '22', '23', '24', '25', '26', '27', '28', '29'] 30 # after init second Serial instance (pyserial)

## close stage

['0', '1', '2', '3', '4', '5', '6', '7', '8', '9', '10', '11', '12', '13', '14', '15', '16', '17', '18', '19', '20', '23', '24', '25', '26', '27', '28', '29'] 28 # called self.stop_redirect_thread()

['0', '1', '2', '3', '4', '5', '6', '7', '8', '9', '10', '11', '12', '13', '14', '16', '23', '24', '25', '26', '27', '28', '29'] 23 # close first serial self.proc.close() 

['0', '1', '2', '3', '4', '5', '6', '7', '8', '9', '10', '11', '12', '13', '14', '16', '23', '24', '25', '26', '27'] 21 # called second self.stop_redirect_thread()

['0', '1', '2', '3', '4', '5', '6', '7', '8', '9', '10', '11', '12', '13', '14', '16'] 16 # close second serial self.proc.close()

['0', '1', '2', '3', '4', '5', '6', '7', '8', '9', '10', '12', '13', '14', '16'] 15 # close io buffered reader?

['0', '1', '2', '3', '4', '5', '6', '7', '8', '9', '10', '12', '13', '16'] 14 #close second buffered reader io buffered reader?

['0', '1', '2', '3', '4', '5', '8', '9', '10', '12', '13', '16'] 12 # close msg query 

['0', '1', '2', '3', '4', '5', '10', '12', '13', '16'] 10 # secon serial close msg query

horw, 2023-10-27:
> It seems like there is a problem with file descriptors '10', '12', '13', '16'. These file descriptors were opened during the init of the pexpect process(_listener.start()).

@JammuKekkonen Hello, in the last [commit](https://github.com/espressif/pytest-embedded/pull/234/commits/d0d735a82425f0f2c2e18a10b9c4bfa76efd31a8), it seems like I fixed the issue above. Did you encounter the same issue after applying your patch?

JammuKekkonen, 2023-10-27:
Hi @horw 

I did see failure with my patch, just postponed further in the test set, the same what you observed. I'll test your PR today to see if the situation improves even further on my setup.

JammuKekkonen, 2023-10-27:
@horw The initial issue of leaking file descriptors seems to have been fixed by #234 !

However, in my setup, the tests will emanate a lot of pytest.PytestUnhandledThreadExceptionWarnings from the place I patched as a starting point (improvement.patch) which I did not see before on my side. Maybe that attempt to empty the queue has to be wrapped in try...except OSError?

  while not self._event_q.empty():
    File "/usr/lib/python3.9/multiprocessing/queues.py", line 129, in empty
      return not self._poll()
    File "/usr/lib/python3.9/multiprocessing/connection.py", line 260, in poll
      self._check_closed()
    File "/usr/lib/python3.9/multiprocessing/connection.py", line 141, in _check_closed
      raise OSError("handle is closed")

horw, 2023-10-31:
What parameters did you use for pytest-embedded when you encountered the error `pytest.PytestUnhandledThreadExceptionWarnings`? Because I didn't get it with your example. 

I will try to search deeper into this problem. The current solution seems like a workaround...the object doesn't get removed properly, can lead to memory leak. I need to take a look at how lazy_fixtures work, because, as you mentioned, without it, there doesn't seem to be a fd/memory leak problem.

JammuKekkonen, 2023-10-31:
I ran the set with just 'pytest', the attached ini file was used. I'm beginning to speculate that your fix could actually solve the whole issue so we might even be able to drop my patch. I'll try to find some time this week to test this.

JammuKekkonen, 2023-11-02:
@horw I reverted my patch and ran the test set and it passed. I think your fix is enough for this.

horw, 2023-11-02:
> @horw I reverted my patch and ran the test set and it passed. I think your fix is enough for this.

@JammuKekkonen, it's great to know, thank you! However, we still need to find the root of the problem before we can merge it.

PR #234 (2023-11-07): fix(serial.py): Pytest-embedded leaks file descriptors when used with pytest-lazy-fixture (RDT-577)
Running pytest with these arguments, along with pytest-lazy-fixture, can lead to a file descriptor error: 'ValueError: filedescriptor out of range in select()'.
-s --count 2 --embedded-services serial --port  /dev/ttyUSB0|/dev/ttyUSB1 --baud=115200|115200

The current commit helps alleviate this error, but if there are a large number of tests, the file descriptor error still exists.
close #232
closed 2023-11-07 by hfudev