espressif/pytest-embedded #238: test: fix unstable `test_detect_port_with_cache`

hfudev, 2023-11-07:
Sometimes the test is unstable. For example, https://github.com/espressif/pytest-embedded/actions/runs/6783180237/job/18436888658

Fails for unrelated changes.

horw, 2023-11-09:
Hello, @hfudev!

Do you mean this type of error? Or is there something else that I need to notice?
**test**: test_teardown_called_for_multi_dut?
```
>           self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
E           FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyUSB100'
```

horw, 2023-11-16:
@hfudev, I've moved the logic to SessionArgs, but I have not removed BuildAppJob because this class has some differences compared to SessionArgs. On the other hand, if you like this solution, perhaps we can find ways to simplify other parts of the code that are related to the args?

horw, 2023-11-17:
I've checked the log, and there were four ports: USB0, USB1, USB2, and ttyS0. USB2 was being used by the esp32s2. However, for the other ports, we encountered the following errors. Therefore, from my perspective, the issue does not seem to be related to the cache, but rather something else.
```
2023-11-07 10:49:11 [dut-1] /dev/ttyUSB0 failed to connect: Failed to connect to ESP32: No serial data received.
2023-11-07 10:49:11 [dut-1] For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
2023-11-07 10:49:11 [dut-1] Serial port /dev/ttyUSB1
2023-11-07 10:49:11 [dut-1] Connecting....
2023-11-07 10:49:11 [dut-1] /dev/ttyUSB1 failed to connect: This chip is ESP32-C3 not ESP32. Wrong --chip argument?
2023-11-07 10:49:11 [dut-1] Serial port /dev/ttyS0
2023-11-07 10:49:11 [dut-1] Connecting........2023-11-07 10:49:13 [dut-0] green...
..........
2023-11-07 10:49:15 [dut-1] /dev/ttyS0 failed to connect: Failed to connect to ESP32: No serial data received.
```

hfudev, 2024-02-08:
thanks for noticing!