espressif/pytest-embedded #257: Support `expect` With `not_matching`

hfudev, 2024-01-08:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

For example

```python
dut.expect('foo', not_matching='bar')
```

Will fail

```
bar foo
```

But pass

```
foo bar
```

check the `pexpect_proc.buffer` after a success match would be a quick solution.

gytxxsy, 2024-01-08:
If possible, would it be more better to allow not_matching to be a list? Such as:
```python

dut.expect('foo', not_matching=[‘A’, 'B', ‘C’]) # The order of A B C is not strictly required
```

hfudev, 2024-01-08:
> If possible, would it be more better to allow not_matching to be a list? Such as:
> 
> ```python
> dut.expect('foo', not_matching=[‘A’, 'B', ‘C’]) # The order of A B C is not strictly required, any one of them will cause expect to fail.
> ```

Hi @gytxxsy thanks for your input! Yes, it should behave like the `pattern` param in all `expect` functions, accepts `str`, `regex`, and list of `str` or `regex`

closed 2024-02-01 by hfudev