espressif/pytest-embedded #303: pytest-embedded-wokwi doesn't work in Windows

urish, 2024-08-18:
Following 1383c2e7b8417839a21bc55d171ec807b090f3d6, the plugin no longer works on windows, since the code uses `pexpect.spawn`, which is not available on Windows.

When using pytest-embedded-wokwi on windows, you get the following error:

```
C:\dev\esp\tools\python_env\idf5.4_py3.12_env\Lib\site-packages\pytest_embedded\plugin.py:490: in wrapper
    res = func(*args, **kwargs)
C:\dev\esp\tools\python_env\idf5.4_py3.12_env\Lib\site-packages\pytest_embedded\plugin.py:1139: in wokwi
    return wokwi_gn(**locals())
C:\dev\esp\tools\python_env\idf5.4_py3.12_env\Lib\site-packages\pytest_embedded\dut_factory.py:475: in wokwi_gn   
    return cls(**_drop_none_kwargs(kwargs))
C:\dev\esp\tools\python_env\idf5.4_py3.12_env\Lib\site-packages\pytest_embedded_wokwi\wokwi_cli.py:62: in __init__
    child = pexpect.spawn('wokwi-cli --help')
E   AttributeError: module 'pexpect' has no attribute 'spawn'
```

urish, 2024-08-18:
For anyone else coming across this issue: right now, the workaround is to use an older version on Windows, e.g. version 1.10.3:

```
pip install pytest_embedded_wokwi==1.10.3
```

hfudev, 2024-08-19:
@urish thanks for reporting. Please have another try on the latest version 1.11.5.

closed 2024-08-19 by hfudev

urish, 2024-08-19:
Works great, thanks!

PR #304 (2024-08-19): fix(wokwi): pexpect.spawn not working on windows. use subprocess instead
closes #303