espressif/pytest-embedded #313: Add a LICENSE file to this project

kbroch-rivosinc, 2024-10-23:
Wondering what the licensing for this project is?  I know many other pytest plugins are MIT or Apache.

kbroch-rivosinc, 2024-10-24:
I should have looked in the pyproject.toml and see the MIT license 
https://github.com/espressif/pytest-embedded/blob/10df30df1e7d9388caabaad015413d046b5fcd51/pytest-embedded/pyproject.toml#L14

I still think it would be useful to get a LICENSE file in since github can't figure out pyproject.toml files

hfudev, 2024-10-24:
Hi @kbroch-rivosinc the LICENSE file is included in each python package. For example https://github.com/espressif/pytest-embedded/blob/main/pytest-embedded-arduino/LICENSE

but yes, you're right, we're not attaching the LICENSE file for the other stuffs, like the examples, and the documentation. I'll create one PR to add the license file. Thank you :)

PR #321 (2025-01-07): chore: add license for the repo
closes #313
closed 2025-01-07 by hfudev