espressif/pytest-embedded #326: [**idf**] Support Skip Marker for ESP-IDF Soc Caps

hfudev, 2024-12-20:
[Soc Caps](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/soc_caps.html) defines hardware capabilities or ESP-IDF apps.

## Ideal Use Case

Support a marker, like `pytest.mark.skip_if_soc("SOC_ADC_SAMPLE_FREQ_THRES_HIGH == 20000")`

Iterate through all supported targets, if the expression is false, then we skip running the test case for this target.

Support iterating through preview targets as well, by `pytest.mark.skip_if_soc("SOC_ADC_SAMPLE_FREQ_THRES_HIGH == 20000", include_preview_targets=True)`

Or iterate over subset of targets, if user specified `targets` as param, like

```python
@pytest.mark.skip_if_soc("SOC_ADC_SAMPLE_FREQ_THRES_HIGH == 20000")
@pytest.mark.parametrization('target', ['esp32', 'esp32s2'], indirect=True)
def test_foo():
    pass
```

here shall only iterate over esp32 and esp32s2

## Implementation Details

the parsing soc caps part could be done by [`idf_build_apps.manifest.if_parser`](https://docs.espressif.com/projects/idf-build-apps/en/latest/references/api/idf_build_apps.manifest.html#idf-build-apps-manifest-package)

the skip test case part shall be done in pytest hook `pytest_collection_modifyitems`

`idf-build-apps` could be added as a dependency for `pytest-embedded-idf`

`SUPPORTED_TARGETS` and `PREVIEW_TARGETS` can also be loaded from `idf_build_apps`. These values are lazy loaded from IDF_PATH.

Make sure everything is lazy-loaded, otherwise it would break user without IDF_PATH.

horw, 2024-12-20:
@hfudev Hi, I can do it, WDYT?

horw, 2024-12-22:
Should we also support the `and` syntax?  
There are two possible variants:

```python
@pytest.mark.custom_skip(
    ("SOC_ADC_SAMPLE_FREQ_THRES_HIGH == 20000", "SUPPORT_A_FEATURE == 1"),
    "SOC_ADC_SAMPLE_FREQ_THRES_HIGH == 10000",
    include_preview_targets=True
)
```

```python
@pytest.mark.custom_skip(
    "SOC_ADC_SAMPLE_FREQ_THRES_HIGH == 20000 and SUPPORT_A_FEATURE == 1",
    "SOC_ADC_SAMPLE_FREQ_THRES_HIGH == 10000",
    include_preview_targets=True
)
```

hfudev, 2024-12-22:
@horw sure. the idf-build-apps [if-clause](https://docs.espressif.com/projects/idf-build-apps/en/latest/references/manifest.html#if-clauses) supports `and` `or` that we may reuse.

horw, 2024-12-23:
The easiest part was `3. Limited Targets` because we can associate our pytest item with the target. However, for `1. Basic` and `2. Preview Targets`, it’s trickier because we don’t know which target it is associated with. So, maybe we should do this only for tests with a target in the `parametrize`?

hfudev, 2024-12-23:
sure it makes sense

PR #327 (2025-01-16): feat: added skip_if_soc marker for idf target

## Description
This PR introduces a `skip_if_soc` marker that helps filter targets based on their `SOC_CAPS.h` attributes.

## Related
Closes #326
closed 2025-01-16 by hfudev