espressif/pytest-embedded #328: mult-dev

Oreobird, 2024-12-27:
Using multi device (2 esp32c3) to test UART send and recv，some time it works, most of the time is fails with error log:
2024-12-27 14:20:02 [dut-0] 
2024-12-27 14:20:02 [dut-0] 
2024-12-27 14:20:02 [dut-0] Press ENTER to see the list of tests.
2024-12-27 14:20:02 [dut-1] Running stub...
2024-12-27 14:20:02 [dut-1] Stub running...
2024-12-27 14:20:02 [dut-1] Changing baud rate to 921600
2024-12-27 14:20:02 [dut-1] Changed.
2024-12-27 14:20:02 [dut-1] Configuring flash size...
2024-12-27 14:20:02 [dut-1] Flash will be erased from 0x00000000 to 0x00005fff...
2024-12-27 14:20:02 [dut-1] Flash will be erased from 0x00008000 to 0x00008fff...
2024-12-27 14:20:02 [dut-1] Flash will be erased from 0x0000d000 to 0x0000efff...
2024-12-27 14:20:02 [dut-1] Flash will be erased from 0x00010000 to 0x0010afff...
2024-12-27 14:20:02 [dut-1] Compressed 20528 bytes to 12606...
Wrote 20528 bytes (12606 compressed) at 0x00000000 in 0.7 seconds (effective 249.1 kbit/s)...
2024-12-27 14:20:03 [dut-1] Hash of data verified.
2024-12-27 14:20:03 [dut-1] Compressed 3072 bytes to 155...
Wrote 3072 bytes (155 compressed) at 0x00008000 in 0.1 seconds (effective 277.0 kbit/s)...
2024-12-27 14:20:03 [dut-1] Hash of data verified.
2024-12-27 14:20:03 [dut-1] Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x0000d000 in 0.2 seconds (effective 376.9 kbit/s)...
2024-12-27 14:20:03 [dut-1] Hash of data verified.
2024-12-27 14:20:03 [dut-1] Compressed 1024992 bytes to 591461...
Wrote 1024992 bytes (591461 compressed) at 0x00010000 in 15.2 seconds (effective 539.7 kbit/s)...
2024-12-27 14:20:22 [dut-1] Hash of data verified.
2024-12-27 14:20:22 [dut-1] 
2024-12-27 14:20:22 [dut-1] Leaving...
2024-12-27 14:20:22 [dut-1] Hard resetting via RTS pin...
2024-12-27 14:20:22 [dut-0] 
2024-12-27 14:20:22 [dut-0] 
2024-12-27 14:20:22 [dut-0] 
2024-12-27 14:20:22 [dut-0] Here's the test menu, pick your combo:
2024-12-27 14:20:22 [dut-0] (1)	"Uart perf2024-12-27 14:20:22 [dut-1] ESP-ROM:esp32c3-api1-20210207
2024-12-27 14:20:22 [dut-1] Build:Feb  7 2021
2024-12-27 14:20:22 [dut-1] rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
2024-12-27 14:20:22 [dut-1] SPIWP:0xee
2024-12-27 14:20:22 [dut-1] mode:DIO, clock div:1
2024-12-27 14:20:22 [dut-1] load:0x3fcd5820,len:0x1708
2024-12-27 14:20:22 [dut-1] load:0x403cc710,len:0x968
2024-12-27 14:20:22 [dut-1] load:0x403ce710,len:0x2f68
2024-12-27 14:20:22 [dut-1] entry 0x403cc710
2024-12-27 14:20:22 [dut-1] I (30) boot: ESP-IDF 3a9c90b6 2nd stage bootloader
2024-12-27 14:20:22 [dut-1] I (30) boot: compile time Dec 27 2024 14:19:15
2024-12-27 14:20:22 [dut-1] I (30) boot: chip revision: v0.4
2024-12-27 14:20:22 [dut-1] I (33 test" [uart_perf][test_env=generic_multi_device][multi_device]
2024-12-27 14:20:22 [dut-0] 	(1)	"master_recv_buf"
2024-12-27 14:20:22 [dut-0] 	(2)	"slave_send_buf"
2024-12-27 14:20:22 [dut-0] 
2024-12-27 14:20:22 [dut-0] Enter test for running.
2024-12-27 14:20:22 [dut-0] 
2024-12-27 14:20:22 [dut-0] 
2024-12-27 14:20:22 [dut-0] 
2024-12-27 14:20:22 [dut-0] Here's the test menu, pick your combo:
2024-12-27 14:20:22 [dut-0] (1)	"Uart  test" [uart_perf][test_env=generic_multi_device][multi_device]
2024-12-27 14:20:22 [dut-0] 	(1)	"master_recv_buf"
2024-12-27 14:20:22 [dut-0] 	(2)	"slave_send_buf"
2024-12-27 14:20:22 [dut-0] 
2024-12-27 14:20:22 [dut-0] Enter test for running.
) boot.esp32c3: SPI Speed      : 80MHz
2024-12-27 14:20:22 [dut-1] I (38) boot.esp32c3: SPI Mode       : DIO
2024-12-27 14:20:22 [dut-1] I (43) boot.esp32c3: SPI Flash Size : 2MB
2024-12-27 14:20:22 [dut-1] I (48) boot: Enabling RNG early entropy source...
2024-12-27 14:20:22 [dut-1] I (53) boot: Partition Table:
2024-12-27 14:20:22 [dut-1] I (57) boot: ## Label            Usage          Type ST Offset   Length
2024-12-27 14:20:22 [dut-1] I (64) boot:  0 nvs              WiFi data        01 02 00009000 00004000
2024-12-27 14:20:22 [dut-1] I (71) boot:  1 otadata          OTA data         01 00 0000d000 00002000
2024-12-27 14:20:22 [dut-1] I (79) boot:  2 phy_init         RF data          01 01 0000f000 00001000
2024-12-27 14:20:22 [dut-1] I (86) boot:  3 ota_0            OTA app          00 10 00010000 0019c800
2024-12-27 14:20:22 [dut-1] I (94) boot:  4 usercfg          WiFi data        01 02 001ad000 00010000ERROR
2024-12-27 14:20:22 [dut-1] I (101) boot:  5 fac_data         WiFi data        01 02 001bd000 00004000
2024-12-27 14:20:22 [dut-1] I (109) boot:  6 log              WiFi data        01 02 001c1000 00010000
2024-12-27 14:20:22 [dut-1] I (116) boot: End of partition table
2024-12-27 14:20:22 [dut-1] I (121) boot: No factory image, trying OTA 0
2024-12-27 14:20:22 [dut-1] I (126) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=23128h (143656) map
2024-12-27 14:20:22 [dut-1] I (157) esp_image: segment 1: paddr=00033150 vaddr=3fc92200 size=02ac4h ( 10948) load
2024-12-27 14:20:22 [dut-1] I (159) esp_image: segment 2: paddr=00035c1c vaddr=40380000 size=0a3fch ( 41980) load
2024-12-27 14:20:22 [dut-1] I (171) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=c25a0h (796064) map

================================================================================================= ERRORS ==================================================================================================
_______________________________________________________________________ ERROR at setup of test_uart[2-defaults-115200] ________________________________________________________________________
../../application/ci_test/perf_test_apps/conftest.py:622: in case_tester
    yield CaseTester(dut, **kwargs)
../../application/ci_test/perf_test_apps/conftest.py:329: in __init__
    super().__init__(dut, **kwargs)
../../application/ci_test/perf_test_apps/conftest.py:170: in __init__
    self.get_test_menu()
../../application/ci_test/perf_test_apps/conftest.py:183: in get_test_menu
    self.test_menu = self.dut.test_menu
../toolchains/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/pytest_embedded_idf/unity_tester.py:243: in test_menu
    self._test_menu = self._parse_test_menu()
../toolchains/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/pytest_embedded_idf/unity_tester.py:126: in _parse_test_menu
    return self._parse_unity_menu_from_str(res.group(1).decode('utf8'))
../toolchains/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/pytest_embedded_idf/unity_tester.py:220: in _parse_unity_menu_from_str
    raise NotImplementedError('Unrecognized test case:', case)
E   NotImplementedError: ('Unrecognized test case:', 'Enter test for running.')
------------------------------------------------------------------------------------------- Captured log setup --------------------------------------------------------------------------------------------
INFO     root:conftest.py:679 find valid binary path: /home/build/esp-idf_v5.1.1
INFO     root:conftest.py:679 find valid binary path: /home/build/esp-idf_v5.1.1
INFO     root:serial.py:121 Target: esp32c3, Port: /dev/ttyUSB1
INFO     root:serial.py:121 Target: esp32c3, Port: /dev/ttyUSB0

horw, 2024-12-27:
Hello! Thank you for your feedback. Can you provide more information? Which command do you use to run this script? Also, could you provide the code to test it (you can attach a link to your repo)?

Oreobird, 2024-12-30:
Works for me to modify _parse_test_menu() in pytest-embedded-idf/unity_tester.py just match the pattern only once：
```
    def _parse_test_menu(
        self,
        ready_line: str = 'Press ENTER to see the list of tests',
        pattern="Here's the test menu, pick your combo:(.+?)Enter test for running.",
        trigger: str = '',
    ) -> t.List[UnittestMenuCase]:
```

closed 2024-12-30 by Oreobird