espressif/pytest-embedded #358: Issues with expect?

evandavey, 2025-05-21:
I am using pytest with ESP-IDF v5.4.1 to run some automated tests.  My test_app is setup to just run     unity_run_menu();

I want to extract the device id for a test report so I have written a test to print it.  This is my pytest to first check the Device ID is printed (before I attempt to write a regex to extract it).  The tests correctly upload and run, I can see the device_id test being selected and the correct device ID being output.  However pytest is unable to parse the output and I just see "Enter next test, or 'enter' to see menu" in the buffer of the error message (like this was the only line expect tried to parse).

I have tried various methods for extracting the full output for parsing without success e.g. dut.read(...) as they don't seem supported.

```
import pytest
from pytest_embedded import Dut
import re

@pytest.mark.supported_targets
@pytest.mark.generic
def test_device_id(dut: Dut) -> None:
    dut.run_all_single_board_cases(name='device_id')
   
    # Wait for the output with a timeout
    dut.expect('Device ID:'), timeout=20)
  
```

evandavey, 2025-05-21:
This appears to be an issues with run_all_single_board_cases consuming all the output.  This works (where 9 is the number of the device_id test in the menu), but is a much less robust approach.

```
@pytest.mark.supported_targets
@pytest.mark.generic
def test_device_id(dut: Dut) -> None:
    dut.expect("Press ENTER to see the list of tests")
    dut.write(( "9\n")
    
    match = dut.expect(r'Device ID: ([0-9A-F]{20})', timeout=20)
    if match and match.group(1):
        device_id = match.group(1).decode('utf-8')
        print(f"Extracted Device ID: {device_id}")
    else:
        pytest.fail("Device ID format incorrect or not captured.")
```

horw, 2025-07-01:
@evandavey Hello, as I understand your issue, it seems like a feature request to make "expect" possible for an entire file, is that correct?
Currently, every `expect` command clears the buffer, so it is not possible to retrieve information from the beginning.